ceccocats/tkDNN #223: plugins for yolov4-csp

chandu1263, 2021-04-23:
@mive93 
I tried running this repository https://github.com/ioir123ju/tkDNN to deserialize the rt engine file using python API.
It is working very well for yolov4 model 
but facing issue while deserializing yolov4-csp
'''
 python darknetTR.py yolo4-csp_fp16_trt700_b1.rt --video=demo/yolo_test.mp4
yolo4-csp_fp16_trt700_b1.rt
New NetworkRT (TensorRT v7)
Float16 support: 1
Int8 support: 1
DLAs: 0
Cant deserialize Plugin
/trt700/python_tkDNN/tkDNN/src/NetworkRT.cpp:821
Aborting...
'''

It would be nice if the plugins are developed for yolov4-csp too.

hlacik, 2021-04-23:
try mine fork  - https://github.com/hlacik/tkdnn , and let me know

chandu1263, 2021-04-24:
@hlacik  Thank you. I will try and let you know. Btw does this support batching?

mive93, 2021-08-19:
@chandu1263  yolov4-csp is supported on tkDNN. 
@bjajoh I currently do not have access to boards to make the tests.