ceccocats/tkDNN #226: 当前版本比之前版本性能差距巨大

dongxuanlb, 2021-05-05:
在adac8576b0faf515ad3f459b1f50fd16cef6d64d版本 xavier使用yolov4可以fp16可以跑到33fps
在最新的版本a638592fc74668471e87ac930b4695ce99dc7d43， 只能跑到10fps

closed 2021-05-16 by ceccocats

ceccocats, 2021-05-16:
thank you, 
it was a compilation issue introduced by @perseusdg during the develop of #218 in this commit 56feb54377c0678e42077fabbf84ce2fc138f4c5

dongxuanlb, 2021-05-26:
Hi, Im still got only 15fps.

dongxuan@xavier:~/workspace/tkDNN/build$ ./demo yolo4_fp16.rt ../demo/yolo_test.mp4 y
detection
yolo4_fp16.rt
New NetworkRT (TensorRT v7.13)
Float16 support: 1
Int8 support: 1
DLAs: 2
create execution context
Input/outputs numbers: 4
input index = 0 -> output index = 3
Data dim: 1 3 416 416 1
Data dim: 1 255 13 13 1
RtBuffer 0   dim: Data dim: 1 3 416 416 1
RtBuffer 1   dim: Data dim: 1 255 52 52 1
RtBuffer 2   dim: Data dim: 1 255 26 26 1
RtBuffer 3   dim: Data dim: 1 255 13 13 1
camera started
^Crequest gateway stop
detection end

Time stats:
Min: 59.4205 ms
Max: 216.611 ms
Avg: 64.0227 ms	15.6195 FPS

commit 6b8ae1e2 (2021-05-16): remove wrong cuda arch that cause big performance gap

fix #226