ceccocats/tkDNN #229: Cmake error

shireen1512, 2021-05-17:
I'm trying to run this on Google Colab, and when i do `cmake ..` i get this error: 

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "kernels" in directory /content/tkDNN

any help would be appreciated.
thank you in advance.

mive93, 2021-05-17:
Hi @shireen1512 
which version of cmake do you have? 
Also, are you sure your CUDA and NVIDIA libraries are correctly installed?

shireen1512, 2021-05-17:
the `cmake --version` gives me _3.12.0_
and `nvcc --version` gives this :
_nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Wed_Jul_22_19:09:09_PDT_2020
Cuda compilation tools, release 11.0, V11.0.221
Build cuda_11.0_bu.TC445_37.28845127_0_

i don't know how to check for the libraries, sorry. i'm new to this.
i just know under usr/lib64-nvidia/ there are files but none have the same kind of name as "CUDA_cublas_device_LIBRARY".
they are like this : 
[image]

the complete error is this :
_-- Found CUDNN: /usr/lib/x86_64-linux-gnu/libcudnn.so
-- Found CUDNN include: /usr/include
CMake Error at CMakeLists.txt:44 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "kernels" in directory /content/tkDNN

-- Configuring incomplete, errors occurred!
See also "/content/tkDNN/build/CMakeFiles/CMakeOutput.log".
See also "/content/tkDNN/build/CMakeFiles/CMakeError.log"._

mive93, 2021-05-17:
It's the cmake version, as written in the README a cmake 3.15 or above is required.

shireen1512, 2021-05-17:
apparently Colab doesn't like me to touch its Cmake... i'll try this on a local machine. thank you.

closed 2021-05-18 by shireen1512