ceccocats/tkDNN #230: Performance Issues on Jetson Xavier NX

MsWik, 2021-05-19:
CV 4.5.0 + CUDA, jetpack 4.5.1. 
Low performance of models like yolo4, yolo4x (lower than in the original darcnet), performance of tiny models is correct.
Also, when I try to convention any model in INT8, I get the following error:
failed to open 
/home/andry/test_jetson_xavier/tkDNN-master/tkDNN/src/Int8BatchStream.cpp:69
Aborting...

perseusdg, 2021-05-19:
Regarding the int8 model conversion ,have you downloaded the dataset and followed the steps mentioned in the Readme? Cause it looks there is a problem with your dataFilePath for image paths 

And about the low performance on Xavier nx ...are you using the current master branch ?

MsWik, 2021-05-19:
FP16 / FP32 dimension models are obtained correctly, the error occurs only when I use INT8. yes, the current master branch is used

MsWik, 2021-05-19:
Yolo 4, yolo 4x performance issue resolved by reinstalling tkDNN, int8 issue not resolved yet.

mive93, 2021-11-25:
@MsWik have you downloaded the dataset as suggested by @perseusdg ? 
Have you exported the needed env variables as explained [here](https://github.com/ceccocats/tkDNN/blob/master/docs/demo.md#int8-inference)?

I tested everything yesterday on a Jetson Xavier AGX and I got no problems.