ceccocats/tkDNN #232: Make fails for pluginsRT

morrolinux, 2021-05-31:
Hi, I'm running Ubuntu 18.04 and installed the dependencies as stated in the readme. 
The `cmake ..` configuration went fine, but `make` fails with:
Any idea about what could be wrong? 
PS: I've got multiple versions of CUDA, CuDNN, TensorRT etc and I'm (supposedly) pointing to the right paths with `$LD_LIBRARY_PATH`.

```
[  1%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o
/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(4): error: incomplete type is not allowed

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(16): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(20): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(24): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(30): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(35): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(38): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(42): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(50): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(54): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(4): error: incomplete type is not allowed

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(16): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(20): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(24): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(30): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(35): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(38): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(42): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(50): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationLogisticRT.h(54): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(4): error: incomplete type is not allowed

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(15): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(19): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(23): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(29): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(34): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(37): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(41): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(49): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationReLUCeilingRT.h(53): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationMishRT.h(4): error: incomplete type is not allowed

/home/morro/tkDNN/include/tkDNN/pluginsRT/ActivationMishRT.h(16): error: member function declared with "override" does not override a base class member
[... 126 lines cut ...]

/home/morro/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(94): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(4): error: incomplete type is not allowed

/home/morro/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(15): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(19): error: member function declared with "override" does not override a base class member

/home/morro/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(23): error: member function declared with "override" does not override a base class member

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00004e3c_00000000-6_normalize.cpp1.ii".
Compilation terminated.
CMake Error at kernels_generated_normalize.cu.o.cmake:280 (message):
  Error generating file
  /home/morro/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_normalize.cu.o

CMakeFiles/kernels.dir/build.make:3439: recipe for target 'CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o' failed
make[2]: *** [CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o] Error 1
CMakeFiles/Makefile2:645: recipe for target 'CMakeFiles/kernels.dir/all' failed
make[1]: *** [CMakeFiles/kernels.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```

ceccocats, 2021-05-31:
What is your version of tensortt, cudnn and CUDA?

morrolinux, 2021-05-31:
TensorRT-6.0.1.5
cudnn 7
CUDA 10.0

mive93, 2021-05-31:
Hi @morrolinux 

check that: 
- your cuDNN and TensorRT versions are aligned with CUDA https://github.com/ceccocats/tkDNN/issues/143#issuecomment-724269903
- you're using a cmake version >= 3.15
- check the TensorRT (NVINFER) path when running ``cmake ..``. If it does not print it, delete the folder and redo it.

morrolinux, 2021-05-31:
Thanks @mive93 , that was it!
I had unaligned versions of cuDNN and TensorRT

closed 2021-05-31 by morrolinux

nightduck, 2023-02-14:
Sorry to resurrect closed issues, but I'm getting the exact same error, but don't have unaligned versions of cuDNN and TensorRT

[image]

Also Ubuntu 18.04. LD_LIBRARY_PATH includes the directories found by cmake.

JaneLin-0409, 2023-07-31:
Hi. @nightduck  Have you solved this problem? I met the same issue...

```
dpkg -l | grep cudnn
ii  libcudnn8                                  8.6.0.166-1+cuda11.4                 arm64        cuDNN runtime libraries
ii  libcudnn8-dev                              8.6.0.166-1+cuda11.4                 arm64        cuDNN development libraries and headers
ii  libcudnn8-samples                          8.6.0.166-1+cuda11.4                 arm64        cuDNN samples
dpkg -l | grep nvinfer
ii  libnvinfer-bin                             8.5.2-1+cuda11.4                     arm64        TensorRT binaries
ii  libnvinfer-dev                             8.5.2-1+cuda11.4                     arm64        TensorRT development libraries and headers
ii  libnvinfer-plugin-dev                      8.5.2-1+cuda11.4                     arm64        TensorRT plugin libraries
ii  libnvinfer-plugin8                         8.5.2-1+cuda11.4                     arm64        TensorRT plugin libraries
ii  libnvinfer-samples                         8.5.2-1+cuda11.4                     all          TensorRT samples
ii  libnvinfer8                                8.5.2-1+cuda11.4                     arm64        TensorRT runtime libraries
ii  python3-libnvinfer                         8.5.2-1+cuda11.4                     arm64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                     8.5.2-1+cuda11.4                     arm64        Python 3 development package for TensorRT

nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Sun_Oct_23_22:16:07_PDT_2022
Cuda compilation tools, release 11.4, V11.4.315
Build cuda_11.4.r11.4/compiler.31964100_0
```
but Ubuntu 20.04

nightduck, 2023-07-31:
@JaneLin-0409 
I was running in a docker container. And after software updates in the background on the host machine, docker was no longer able to use CUDA drivers. It was solved with a reboot of the host machine.

It's also important to make sure the version of libcudnn on your docker machine is compatible with the Nvidia drivers on your host machine. CUDA 11.4 requires driver 450 or higher.