ceccocats/tkDNN #233: about windows

Hipeppa, 2021-06-01:
when i use windows version,i meet a mistake, after ninja -j4, it can not success, it output like ：
`[1/117] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_leaky.cu.obj
FAILED: CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_leaky.cu.obj
cmd.exe /C "cd /D D:\learn\tkDNN_liu\tkDNN\build\CMakeFiles\kernels.dir\src\kernels && "D:\Program Files\CMake\bin\cmake.exe" -E make_directory D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/. && "D:\Program Files\CMake\bin\cmake.exe" -D verbose:BOOL=OFF -D build_configuration:STRING=Release -D generated_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_leaky.cu.obj -D generated_cubin_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_leaky.cu.obj.cubin.txt -P D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_leaky.cu.obj.Release.cmake"
nvcc fatal   : Host compiler targets unsupported OS.
CMake Error at kernels_generated_activation_leaky.cu.obj.Release.cmake:219 (message):
  Error generating
  D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_leaky.cu.obj

[2/117] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_mish.cu.obj
FAILED: CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_mish.cu.obj
cmd.exe /C "cd /D D:\learn\tkDNN_liu\tkDNN\build\CMakeFiles\kernels.dir\src\kernels && "D:\Program Files\CMake\bin\cmake.exe" -E make_directory D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/. && "D:\Program Files\CMake\bin\cmake.exe" -D verbose:BOOL=OFF -D build_configuration:STRING=Release -D generated_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_mish.cu.obj -D generated_cubin_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_mish.cu.obj.cubin.txt -P D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_mish.cu.obj.Release.cmake"
nvcc fatal   : Host compiler targets unsupported OS.
CMake Error at kernels_generated_activation_mish.cu.obj.Release.cmake:219 (message):
  Error generating
  D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_mish.cu.obj

[3/117] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_logistic.cu.obj
FAILED: CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_logistic.cu.obj
cmd.exe /C "cd /D D:\learn\tkDNN_liu\tkDNN\build\CMakeFiles\kernels.dir\src\kernels && "D:\Program Files\CMake\bin\cmake.exe" -E make_directory D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/. && "D:\Program Files\CMake\bin\cmake.exe" -D verbose:BOOL=OFF -D build_configuration:STRING=Release -D generated_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_logistic.cu.obj -D generated_cubin_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_logistic.cu.obj.cubin.txt -P D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_logistic.cu.obj.Release.cmake"
nvcc fatal   : Host compiler targets unsupported OS.
CMake Error at kernels_generated_activation_logistic.cu.obj.Release.cmake:219 (message):
  Error generating
  D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_logistic.cu.obj

[4/117] Building NVCC (Device) object CMakeFiles/kernels.d...c/kernels/kernels_generated_activation_relu_ceiling.cu.obj
FAILED: CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_relu_ceiling.cu.obj
cmd.exe /C "cd /D D:\learn\tkDNN_liu\tkDNN\build\CMakeFiles\kernels.dir\src\kernels && "D:\Program Files\CMake\bin\cmake.exe" -E make_directory D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/. && "D:\Program Files\CMake\bin\cmake.exe" -D verbose:BOOL=OFF -D build_configuration:STRING=Release -D generated_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_relu_ceiling.cu.obj -D generated_cubin_file:STRING=D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_relu_ceiling.cu.obj.cubin.txt -P D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_relu_ceiling.cu.obj.Release.cmake"
nvcc fatal   : Host compiler targets unsupported OS.
CMake Error at kernels_generated_activation_relu_ceiling.cu.obj.Release.cmake:219 (message):
  Error generating
  D:/learn/tkDNN_liu/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_relu_ceiling.cu.obj

ninja: build stopped: subcommand failed.`

what problem maybe

Hipeppa, 2021-06-01:
my msvc is 16.10.0, my cuda is 10.2, mu cudnn is 8.0.2.3

perseusdg, 2021-06-01:
If you go through your error you will find this nvcc error "nvcc fatal : Host compiler targets unsupported OS." there could be a couple of a reasons for it : 
1)You could have cl.exe of a different architecture like x86 instead of amd64 in your path,so trying using the x64 native command prompt or use cl.exe from a path similar to this "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/HostX64/x64/cl.exe" (just replace it with the appropriate msvc version)

2)if the error persists even after doing the above changes then the issue could be that 16.10 is an unsupported compiler version for nvcc ,in which case it would be better if you chose a version like 16.7/16.9 ,if however 16.10 is a must for you then you could add this flag (--allow-unsupported-compiler) to the nvcc flags line in CMakeLists.txt

Hipeppa, 2021-06-01:
> If you go through your error you will find this nvcc error "nvcc fatal : Host compiler targets unsupported OS." there could be a couple of a reasons for it :
> 1)You could have cl.exe of a different architecture like x86 instead of amd64 in your path,so try using the x64 native command prompt or use cl.exe from a path similar to this "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/HostX64/x64/cl.exe" (just replace it with the appropriate msvc version)
> 
> 2)if the error persists even after doing the above changes then the issue could be that 16.10 is an unsupported compiler version for nvcc ,in which case it would be better if you chose a version like 16.7/16.9 ,if however 16.10 is a must for you then you could add this flag (--allow-unsupported-compiler) to the nvcc flags line in CMakeLists.txt

thanks you, i will try, if i have answer, i will telll you

perseusdg, 2021-06-06:
Hey I just wanted to let you know I have been able to compile and use tkdnn properly with msvc 16.10 .My configuration is cuda11.1 + cuda 8.1.1 + TensorRt 7.2.3

Hipeppa, 2021-06-06:
> Hey I just wanted to let you know I have been able to compile and use tkdnn properly with msvc 16.10 .My configuration is cuda11.1 + cuda 8.1.1 + TensorRt 7.2.3

thanks you ,i have pass this step, and i find the problem, the cmake in my computer use gcc and g++ by default, so before this step, i use set CC=cl, it succeed, now i meet a new problem, it output:
D:\learn\tkDNN_liu\mytkDNN\include\tkDNN\NetworkRT.h(8): fatal error C1083: 无法打开包括文件: “NvInfer.h”: No such file or directory
but when i  cmake, it output:
-- Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2 (found suitable version "10.2", minimum required is "9.0")
-- Found CUDNN: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2/lib/x64/cudnn.lib
-- Found CUDNN include: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2/include
-- Found NVINFER: D:/TensorRT-7.0.0.11/lib/nvinfer.lib
-- Found NVINFER include: D:/TensorRT-7.0.0.11/include
-- Found CUDNN: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2/lib/x64/cudnn.lib
-- Found OpenCV: C:/opt/x64-windows (found version "4.5.1")
it prove the tensorrt has no problem in my computer, so why it can not find NvInfer.h.also, i had use tensorrt in my computer, it can use normally.
Sorry to disturb you.

perseusdg, 2021-06-06:
I am not exactly sure why it cant find NvInfer.h ,I would suggest that you add NvInfer.h to : C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2/include as that configuration seems to work

Hipeppa, 2021-06-07:
> I am not exactly sure why it cant find NvInfer.h ,I would suggest that you add NvInfer.h to : C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.2/include as that configuration seems to work

thanks you, do as you say, i succeed.

closed 2021-06-07 by Hipeppa

Hezhexi2002, 2021-08-28:
> @Hezhexi2002 refer to this comment [#252 (comment)](https://github.com/ceccocats/tkDNN/issues/252#issuecomment-897738220)

Thanks a lot,I have tried to use the master branch you provided and built it again,However,now I met a new error:
[image]
It seems like it is caused by my Eigen path,but my cmake result is correct:
[image]
It can detect my own Eigen properly,so waht's the reason for that?and here is my another error I met yesterday [https://github.com/ceccocats/tkDNN/issues/259](url),hope you can see it

Hezhexi2002, 2021-08-28:
> Thanks a lot,I have tried to use the master branch you provided and built it again,However,now I met a new error:
> [image]
> It seems like it is caused by my Eigen path,but my cmake result is correct: [...]

Sorry to bother you again,It should be my Eigen path because the Eigen DIR in the cmake result is not my own path,my correct path should be:
[image]
but when I set the path in the CMaKeLists.txt and cmake it again the path is still wrong,I used to set the OpenCV_DIR into my own opencv path and cmake can detect it correctly but why it didn't work for Eigen?here is my CMaKLists.txt:
[CMakeLists.txt](https://github.com/ceccocats/tkDNN/files/7070058/CMakeLists.txt)

Hipeppa, 2021-08-28:
> t when I set the path in the CMaKeLists.txt and cmake it again the path is still wrong,I used to set the OpenCV_DIR into my own opencv path and cmak

加我微信13996295096，我应该知道是什么问题了

Hipeppa, 2021-08-28:
> Sorry to bother you again,It should be my Eigen path because the Eigen DIR in the cmake result is not my own path,my correct path should be:
> [image]
> but when I set the path in the CMaKeLists.txt and [...]

加我微信15996295096，我应该知道是什么问题

perseusdg, 2021-08-28:
what is the version of eigen3 you are using ?

Hipeppa, 2021-08-28:
> what is the version of eigen3 you are using ?

i use the latest version, i have no worry, using vcpkg, it will install the latest version automatically

Hezhexi2002, 2021-08-28:
> what is the version of eigen3 you are using ?

3.39

Hezhexi2002, 2021-08-28:
> > what is the version of eigen3 you are using ?
> 
> 3.39，I used to compile it from the source code using visual studio

Hezhexi2002, 2021-08-28:
> > what is the version of eigen3 you are using ?
> 
> i use the latest version, i have no worry, using vcpkg, it will install the latest version automatically

Does it mean it is causeed by eigen version?