ceccocats/tkDNN #234: Error with make

pereayats, 2021-06-03:
I get several errors like this at the last step of the compilation:

`/root/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(53): error: member function declared with "override" does not override a base class memb
er

/root/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(56): error: member function declared with "override" does not override a base class memb
er

/root/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(60): error: member function declared with "override" does not override a base class memb
er

/root/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(90): error: member function declared with "override" does not override a base class memb
er

/root/tkDNN/include/tkDNN/pluginsRT/YoloRT.h(94): error: member function declared with "override" does not override a base class memb
er

/root/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(4): error: incomplete type is not allowed

/root/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(15): error: member function declared with "override" does not override a base class
member

/root/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(19): error: member function declared with "override" does not override a base class
member

/root/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(23): error: member function declared with "override" does not override a base class
member

Error limit reached.
100 errors detected in the compilation of "/root/tkDNN/src/kernels/normalize.cu".
Compilation terminated.
CMake Error at kernels_generated_normalize.cu.o.cmake:279 (message):
  Error generating file
  /root/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_normalize.cu.o

make[2]: *** [CMakeFiles/kernels.dir/build.make:3656: CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:641: CMakeFiles/kernels.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

All the dependencies are installed and the other steps look fine. Has anyone an idea of what it might be happening?

casperbh96, 2021-06-05:
I got the same when I tried with TensorRT 8.0 EA, CUDNN 8.2.0 and CUDA 11.3.

What dependencies do you have installed?

Also, can we fix this?

daynauth, 2021-06-06:
it might be a c++ version error. are you compiling c++ 17, 14 or 11?

perseusdg, 2021-06-06:
I believe TensorRT 8 isn't supported at the moment , according to the [release notes](https://docs.nvidia.com/deeplearning/tensorrt/release-notes/tensorrt-8.html#tensorrt-8) in the Deprecated and Removed section, you will notice IPlugin and IPluginFactory have been removed ,these two interfaces are currently being used in tkDNN , so at the moment I believe the maximum supported version is TensorRT 7.2.3.If TensorRt 8 is a must for you then you can check the plugin [migration guide](https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#migrating-plugins-6x-7x-to-8x) to migrate from v6/v7 to v8 or wait till tkDNN supports it .

casperbh96, 2021-06-06:
To elaborate; I just bought a laptop that is going to function as a development machine for Darknet models that is converted to static TRT models with tkDNN. It has a Ryzen 5700H and RTX 3070; this combination of new hardware is only supported from Ubuntu 20.04, nvidia driver 460, CUDA 11.1. Now, TensorRT only supports Ubuntu 20.04 from TensorRT 8.0.

@perseusdg do you have any idea how hard this would be to implement in tkDNN? I have ~4 years of coding in Python, Java, C# but no real prior experience in C++. However, I could pick it up and try to replace the interfaces.

perseusdg, 2021-06-06:
I have been using TensorRt 7.2.3 with cuda 11.1 on my Ubuntu (pop os) 20.04 machine for quite sometime now ,i use the ubuntu18.04 tar packages on 20.04 and it has worked pretty amazingly for me.
If you are uncomfortable with this then you could try using [docker](https://github.com/ceccocats/tkDNN/tree/master/docker) .
When it comes to replacing the interfaces,it shouldn't take much time once you have a good idea about TensorRt ,how custom plugins are implemented (refer[ documentation](https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#add_custom_layer)) and a basic understanding of tkDNN's code.

Personally speaking I would recommend going with option 1 or  2 as the current release of TensorRt 8 is an ea and has quite a number of known issues mentioned in the release notes,if performance or stability isn't an issue then going ahead with replacing interfaces sounds good.

casperbh96, 2021-06-06:
> I have been using TensorRt 7.2.3 with cuda 11.1 on my Ubuntu (pop os) 20.04 machine for quite sometime now ,i use the ubuntu18.04 tar packages on 20.04 and it has worked pretty amazingly for me.
> If [...]

Thank you for your answer. It seems that there should be no issue with TensorRT (yet). I was not aware that you could use the Ubuntu 18.04 version.

casperbh96, 2021-06-06:
I can confirm that TensorRT + tKDNN only works with TensorRT <= 7.2.3.

This issue should be closed - install TensorRT 7.2.3 or below to fix any issues.

pereayats, 2021-06-07:
I have installed TensorRT 7.2.3 and I get the same errors. I have TensorRT 7.2.3 Ubuntu 18.04 version, CUDA 11.0 and CUDNN 8.2. Anyone has any idea of what could be happening?

perseusdg, 2021-06-07:
Try using cudnn 8.1 with TensorRt 7.2.3

casperbh96, 2021-06-08:
My specific setup right now is

- Ubuntu 20.04
- CUDA 11.1
- CUDNN 8.0.5.39
- TensorRT 7.2.2.3

I would suggest trying this configuration as they are all compatible. Here is my installation script (note that you need to download the TensorRT deb package for the TensorRT part to work):

```bash
#!/bin/bash

# available version guides:
# CUDA versions: apt list -a cuda
# CUDNN versions: apt list -a libcudnn8

remove_previous=false
OS=ubuntu2004
cuda_version=11.1
cudnn_version=8.0.5.39
tensorrt=7.2.2

# remove previous packages
if [ "$remove_previous" = true ] ; then
    sudo apt purge cuda*
    sudo apt purge nv-tensorrt*
    sudo apt purge libcudnn*
fi

# install CUDA
wget https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/cuda-${OS}.pin
sudo mv cuda-${OS}.pin /etc/apt/preferences.d/cuda-repository-pin-600
sudo apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/7fa2af80.pub
sudo add-apt-repository "deb http://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/ /"
sudo apt-get update
sudo apt-get -y install ${cuda_version}

# install CuDNN
wget https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/cuda-${OS}.pin 

sudo mv cuda-${OS}.pin /etc/apt/preferences.d/cuda-repository-pin-600
sudo apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/7fa2af80.pub
sudo add-apt-repository "deb https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/ /"
sudo apt-get update

sudo apt-get install -y libcudnn8=${cudnn_version}-1+cuda${cuda_version}
sudo apt-get install -y libcudnn8-dev=${cudnn_version}-1+cuda${cuda_version}

# install TensorRT (note: ubuntu 18.04 version works for 20.04)
sudo dpkg -i nv-tensorrt-repo-ubuntu1804-cuda11.1-trt7.2.2.3-ga-20201211_1-1_amd64.deb
sudo apt-key add /var/nv-tensorrt-repo-cuda11.1-trt7.2.2.3-ga-20201211/7fa2af80.pub
sudo apt-get update
sudo aptitude install tensorrt # option 2 works
```

pereayats, 2021-06-10:
> My specific setup right now is
> 
> * Ubuntu 20.04
> * CUDA 11.1
> * CUDNN 8.0.5.39
> * TensorRT 7.2.2.3
> 
> I would suggest trying this configuration as they are all compatible. Here is my installation script (note that you need to download the TensorRT deb package for the TensorRT part to work):
> 
> ```shell
> #!/bin/bash
> 
> # available version guides:
> # CUDA versions: apt list -a cuda
> # CUDNN versions: apt list -a libcudnn8
> 
> remove_previous=false
> OS=ubuntu2004
> cuda_version=11.1
> cudnn_version=8.0.5.39
> tensorrt=7.2.2
> 
> # remove previous packages
> if [ "$remove_previous" = true ] ; then
>     sudo apt purge cuda*
>     sudo apt purge nv-tensorrt*
>     sudo apt purge libcudnn*
> fi
> 
> # install CUDA
> wget https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/cuda-${OS}.pin
> sudo mv cuda-${OS}.pin /etc/apt/preferences.d/cuda-repository-pin-600
> sudo apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/7fa2af80.pub
> sudo add-apt-repository "deb http://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/ /"
> sudo apt-get update
> sudo apt-get -y install ${cuda_version}
> 
> # install CuDNN
> wget https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/cuda-${OS}.pin 
> 
> sudo mv cuda-${OS}.pin /etc/apt/preferences.d/cuda-repository-pin-600
> sudo apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/7fa2af80.pub
> sudo add-apt-repository "deb https://developer.download.nvidia.com/compute/cuda/repos/${OS}/x86_64/ /"
> sudo apt-get update
> 
> sudo apt-get install -y libcudnn8=${cudnn_version}-1+cuda${cuda_version}
> sudo apt-get install -y libcudnn8-dev=${cudnn_version}-1+cuda${cuda_version}
> 
> # install TensorRT (note: ubuntu 18.04 version works for 20.04)
> sudo dpkg -i nv-tensorrt-repo-ubuntu1804-cuda11.1-trt7.2.2.3-ga-20201211_1-1_amd64.deb
> sudo apt-key add /var/nv-tensorrt-repo-cuda11.1-trt7.2.2.3-ga-20201211/7fa2af80.pub
> sudo apt-get update
> sudo aptitude install tensorrt # option 2 works
> ```

Thank you so much! That worked perfectly

closed 2021-06-10 by pereayats