ceccocats/tkDNN #236: OpenCV error when demoing yolo4_tiny

pereayats, 2021-06-09:
I have been trying to run the demo file for yolo4_tiny and I get the following OpenCV error:

`Unable to init server: Could not connect: Connection refused
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.3-pre) /root/opencv/modules/highgui/src/window_gtk.cpp:635: error: (-2:Unspecified error) Can't initialize GTK
backend in function 'cvInitSystem'`

I have seen online that I might be an issue of X11 forwarding and trying with the -X flag for ssh to the VM might solve it, but it still does not work. Anyone has an idea of what it might be the issue? Thanks!

pereayats, 2021-06-10:
Actually I found the solution thanks to this post https://seanhsu.xyz/?p=65. However, you have to use the XQuartz terminal

closed 2021-06-10 by pereayats

Pittmann-XIE, 2023-12-10:
Hello, the link is no longer valid. Do you still remember how to fix that?