ceccocats/tkDNN #246: compile segment error

toplinuxsir, 2021-07-19:
Compile with gcc-10, tensorrt version 8.0.1 
The error:
```

[  8%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_deformable_conv.cu.o                                    
/usr/include/c++/10/chrono: In substitution of ‘template<class _Rep, class _Period> template<class _Period2> using __is_harmonic = std::__bool_con
stant<(std::ratio<((_Period2::num / std::chrono::duration<_Rep, _Period>::_S_gcd(_Period2::num, _Period::num)) * (_Period::den / std::chrono::dura
tion<_Rep, _Period>::_S_gcd(_Period2::den, _Period::den))), ((_Period2::den / std::chrono::duration<_Rep, _Period>::_S_gcd(_Period2::den, _Period:
:den)) * (_Period::num / std::chrono::duration<_Rep, _Period>::_S_gcd(_Period2::num, _Period::num)))>::den == 1)> [with _Period2 = _Period2; _Rep 
= _Rep; _Period = _Period]’:                                                                                                                      
/usr/include/c++/10/chrono:473:154:   required from here                                                                                          
/usr/include/c++/10/chrono:428:27: internal compiler error: Segment Fault                                                                                
  428 |  _S_gcd(intmax_t __m, intmax_t __n) noexcept                                                                                              
      |                           ^~~~~~                                                                                                          
Please submit a full bug report,                                                                                                                  
with preprocessed source if appropriate.                                                                                                          
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.                                                                                  
CMake Error at kernels_generated_activation_elu.cu.o.cmake:280 (message):                                                                         
  Error generating file                                                                                                                           
  /home/lhs/soft/tkdnn/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_activation_elu.cu.o 
```

perseusdg, 2021-07-19:
Firstly ,tkdnn doesn't support tensorrt8 yet .Secondly ,if you are using gcc 10.3 ,it likely won't work there have been issues reported on the gcc forum about gcc 10.3 not being able to compile a simple cuda program and Nvidia supports only till gcc 10.2 ,if gcc 10.3 is a must for you then you could try one of the patches mentioned here https://github.com/NVIDIA/nccl/issues/494

toplinuxsir, 2021-07-20:
@perseusdg Thanks! I will test again.

mive93, 2021-11-24:
TensorRT8 is now supported on [tensorrt8 branch](https://github.com/ceccocats/tkDNN/tree/tensorrt8).
Every model and data type is properly working.

closed 2021-11-24 by mive93