ceccocats/tkDNN #248: low fps in AGX XAVIER

AlanNewImage, 2021-07-26:
Hi:

I test yolo4tiny in AGX use newest repo ,only achieved about 120fps.
launched jetson_clocks
test by test_rtinference
test in two diffrent AGX Xavier and have the same result

Output Buffers: 2
Output Buffers: 2
Output Buffers: 2
Output Buffers: 2
avg: 8.43752

AlanNewImage, 2021-07-26:
Output Buffers: 3
Output Buffers: 3
Output Buffers: 3
Output Buffers: 3
avg: 37.9915
this is yolo4_berkely_fp16,only 26fps
download weights from this repo, and convert model to .rt by test_yolo4_berkeley

sapienzadavide, 2021-07-27:
Hi @AlanNewImage 
What is the commit/branch you are working on? 
Thank you

AlanNewImage, 2021-07-28:
branch master

I get 244fps for yolo4tiny_fp16 based the newest commit, but only 29fps for yolo4_berkeley_fp16.

sapienzadavide, 2021-07-28:
What is the NV power mode? (try MAXN)

AlanNewImage, 2021-07-29:
Thanks for your reply;
CUDA: 10.2
TensorRT: 7.1.3
cudnn: 8.0.0.180

AlanNewImage, 2021-07-30:
Thank you very much!
I get 335 fps after set nvpmodel to 0 .
WOW,it's amazing.
Thanks again.

closed 2021-08-06 by AlanNewImage