ceccocats/tkDNN #250: make error

winterxx, 2021-07-29:
when make a error in centos7:
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::__throw_out_of_range_fmt(char const*, ...)@GLIBCXX_3.4.20'                                                                            
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_notify_all(unsigned int*)@GLIBCXX_3.4.21'                                                      
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::runtime_error::runtime_error(char const*)@GLIBCXX_3.4.21'                                                                             
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::out_of_range::out_of_range(char const*)@GLIBCXX_3.4.21'                                                                               
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::__atomic_futex_unsigned_base::_M_futex_wait_until(unsigned int*, unsigned int, bool, std::chrono::duration<long, std::ratio<1l, 1l> >,
 std::chrono::duration<long, std::ratio<1l, 1000000000l> >)@GLIBCXX_3.4.21'                                                                                                                                  
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::thread::_M_start_thread(std::shared_ptr<std::thread::_Impl_base>, void (*)())@GLIBCXX_3.4.21'                                         
/usr/local/cuda-10.0/lib64/libnvinfer.so: undefined reference to `std::__future_base::_State_baseV2::_Make_ready::_M_set()@GLIBCXX_3.4.21'                                                                   
collect2: error: ld returned 1 exit status                                                                                                                                                                   
make[2]: *** [demo] Error 1                                                                                                                                                                                  
make[1]: *** [CMakeFiles/demo.dir/all] Error 2                                                                                                                                                               
make: *** [all] Error 2

mive93, 2021-08-19:
Have you correctly installed tensorRT for CUDA10.0?

mive93, 2021-11-25:
Closing for inactivity

closed 2021-11-25 by mive93