ceccocats/tkDNN #251: make error

ariharasudhanm, 2021-08-04:
Error limit reached.
100 errors detected in the compilation of "/home/inchstone/tkDNN/src/kernels/normalize.cu".
Compilation terminated.
CMake Error at kernels_generated_normalize.cu.o.cmake:280 (message):
  Error generating file
  /home/inchstone/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_normalize.cu.o

make[2]: *** [CMakeFiles/kernels.dir/build.make:3732: CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:90

mive93, 2021-08-19:
Can you give further details? How did you get this error? On which board/OS?

ariharasudhanm, 2021-08-20:
There were some problems with cuda and CudNN, Tensorrt versions so now it's
perfectly fine after installing the correct matching versions of all three.

mive93, 2021-08-20:
Ok cool, then I'm closing the issue.

closed 2021-08-20 by mive93

sleepingsaint, 2021-10-11:
I am getting the same error, can anyone help me with this issue.
[image]

[image]

I added the screenshots of the tensorrt, cudnn and, cuda version

Any help is very much appreciated. Thanks in advance :)

perseusdg, 2021-10-11:
tensorrt-8 isn't supported at the moment,it is a work in progress ,the latest supported tensorrt version by tkDNN is 7.x

perseusdg, 2021-10-16:
my guess is that you aren't linking opencv libraries properly