ceccocats/tkDNN #252: ninja ERROR

B1ackg0d, 2021-08-12:
ninja -j4
[1/49] Building CXX object CMakeFiles\tkDNN.dir\src\CenterTrack.cpp.obj
FAILED: CMakeFiles/tkDNN.dir/src/CenterTrack.cpp.obj
C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DTKDNN_PATH=\"D:/tkDNN-master1/tkDNN-master\" -DtkDNN_EXPORTS -I..\include\tkDNN -I"E:\CUDA\CUDA Development\include" -IC:\opt\x64-windows\include\eigen3 -I..\include -IC:\opt\x64-windows\include /O2 /FS /EHsc -DOPENCV /O2 /Ob2 /DNDEBUG -MD /showIncludes /FoCMakeFiles\tkDNN.dir\src\CenterTrack.cpp.obj /FdCMakeFiles\tkDNN.dir\ /FS -c ..\src\CenterTrack.cpp
..\src\CenterTrack.cpp(311): warning C4838: 从“int”转换到“float”需要收缩转换
..\src\CenterTrack.cpp(311): warning C4838: 从“int”转换到“float”需要收缩转换
..\src\CenterTrack.cpp(472): error C2131: 表达式的计算结果不是常数
..\src\CenterTrack.cpp(472): note: 因读取超过生命周期的变量而失败
..\src\CenterTrack.cpp(472): note: 请参见“this”的用法
..\src\CenterTrack.cpp(473): error C2131: 表达式的计算结果不是常数
..\src\CenterTrack.cpp(472): note: 因读取超过生命周期的变量而失败
..\src\CenterTrack.cpp(472): note: 请参见“this”的用法
..\src\CenterTrack.cpp(475): error C3863: 不可指定数组类型“bool [sl]”
..\src\CenterTrack.cpp(476): error C2131: 表达式的计算结果不是常数
..\src\CenterTrack.cpp(476): note: 因读取超过生命周期的变量而失败
..\src\CenterTrack.cpp(476): note: 请参见“this”的用法
..\src\CenterTrack.cpp(478): error C3863: 不可指定数组类型“bool ['函数']”
..\src\CenterTrack.cpp(481): error C3863: 不可指定数组类型“bool ['函数']”
..\src\CenterTrack.cpp(483): error C3863: 不可指定数组类型“bool [sl]”
ninja: build stopped: subcommand failed.

D:\tkDNN-master1\tkDNN-master\build>ninja -C out/Default -v
ninja: fatal: chdir to 'out/Default' - No such file or directory
ninja: Entering directory `out/Default'

perseusdg, 2021-08-12:
If you are taking about running the segmentation network(the network has some caveats on linux too,as mentioned in the docs it runs with cuda 10.2 only) then yeah this problem cant be solved yet ,if you are talking about the performance of certain networks (like centertrack) when compared to linux,i think the issue boils down to wddm mode ,maybe someone with a card than can run in tcc mode can achieve a similar performance to running it on linux

B1ackg0d, 2021-08-12:
Sorry, I still don't understand what you mean. In fact, I am just a beginner. I just want to successfully compile tkdnn, try to run the demo. But using the command ninja, it reported a lot of compilation errors (many syntax) in cpp.
My msvc is 16.1, will this affect it? But I see that others can run successfully.

perseusdg, 2021-08-12:
In short use the master branch on this fork https://github.com/perseusdg/tkDNN (it will run every network except the segmentation(seg_demo) ones )  to use it on windows and I am not sure about msvc 16.1 ,I tested it on 16.7,16.9 and 16.10 .If you want to stick to this repo and use it on windows ,check out the repo at this point https://github.com/ceccocats/tkDNN/tree/c306b368608893e92925bf143e7cf14f19525aeb (this works with msvc 16.7 and 16.9)

B1ackg0d, 2021-08-12:
Sorry I made a mistake, my msvc is 16.11.
[image]
Is this the msvc version?Or something like this？
[image]

perseusdg, 2021-08-12:
Yeah this version should work

B1ackg0d, 2021-08-13:
Thanks for your help.The branch you posted can be used.

closed 2021-08-13 by B1ackg0d

grandprixgp, 2021-08-20:
@B1ackg0d funny running into you here :)

@perseusdg Just to clarify, I assume the newly released ShelfNet support does not function on Windows? I was just in the process of implementing this (Windows) and stumbled upon this issue when having build issues.

Do you have any idea what the timeline is looking like for proper ShelfNet on tkDNN on Windows support?

perseusdg, 2021-08-20:
At the moment shelfnet doesn't work on windows and about the timeline,I am not really sure about it.