ceccocats/tkDNN #253: int8.rt unable to generate

B1ackg0d, 2021-08-13:
I want to get yolo4tiny_int8.rt or other version of yolo4's int8.rt. I use test_all_tests.sh, but an error is reported.
```
failed to open ../demo/COCO_val2017/all_images.txt
..\src\Int8BatchStream.cpp:69
Aborting...
unable to read serialRT file
..\tests\test_rtinference\rtinference.cpp:9
Aborting...
```

perseusdg, 2021-08-13:
Did you download COCO_val2017/BDD dataset using either the download_validation.sh(for linux) or download_validation,py(for windows) before running test_all_tests.sh ,also I would recommend against doing this as it will take a considerable amount of time to complete  (as it has to download all weights and generate the RT file for each and everyone of them) ,it might be better to follow the INT8 method mentioned in the docs

B1ackg0d, 2021-08-13:
I get it.Thx!

closed 2021-08-13 by B1ackg0d

B1ackg0d, 2021-08-13:
download_validation.py is so slow, is there any better way?