ceccocats/tkDNN #254: 'darknet.exe export ' doesn't work in Windows version

B1ackg0d, 2021-08-14:
I want to use darknet to export weights on windows, and then use tkdnn-windows to generate rt files.
It seems that darknet does not have this feature on Windows.Do i have a better way to export weights in Windows？
Could you help me?Thx!

perseusdg, 2021-08-14:
you could use a linux vm or wsl to export the weights

B1ackg0d, 2021-08-15:
[image]
Hi, perseusdg.How to fix Error MSB372? I changed the `compute_61,sm_61`,bur it doesn't work.

[image]
[image]

closed 2021-08-16 by B1ackg0d