ceccocats/tkDNN #256: After generating .rt, there is no check boxes when using demo.exe to test

B1ackg0d, 2021-08-19:
After the .rt is generated, the display is as follows：
[image]

mive93, 2021-08-19:
It seems like the debug file are missing. 
Does the demo work? 
Which steps did you follow to reach that result?

B1ackg0d, 2021-08-19:
Hi,mive93!I thought the debug file was not necessary, so I didn’t care about it.
I first used darknet to export the weights. Then I changed the names and cfg files under tkdnn/test/darknet. Then use test_yolo4.exe to generate the rt file, it is as shown in the figure above. Then I run demo.exe to test rt file, and I find that there is no check boxes.

mive93, 2021-08-19:
You should export the debug and check if the conversion works fine.

B1ackg0d, 2021-08-19:
This is what it says in the document，‘if you need debug export of every layer ouput you MUST compile in CPU mode’,I’m not sure how to compile with cpu, so I didn’t use.

mive93, 2021-08-19:
It's quite easy, just leave ```GPU=0``` in darknet makefile

B1ackg0d, 2021-08-19:
Thank you very much! I will try it. After modifying it, recompiling, and then using export, will there be a debug file?

mive93, 2021-08-19:
A debug folder. Just create the debug and layers folder in advance.

B1ackg0d, 2021-08-19:
Thank you for your reply! I have another question. Do the parameters in the training cfg, the exported cfg and the cfg used by tkdnn all need to be consistent?Especially the size of batchsize.

mive93, 2021-08-20:
Yes, they need to be consistent, but the batch size of that config (only used for training) is not used in tkDNN.

B1ackg0d, 2021-08-20:
> It's quite easy, just leave `GPU=0` in darknet makefile
If I use VS2019 to compile darknet, can I just use this sln directly?
[image]

closed 2021-08-22 by B1ackg0d