ceccocats/tkDNN #257: Support Jetpack 4.6

rodja, 2021-08-19:
tkDNN seems not to be able to compile on Jetpack 4.6 (tested on Jetson Nano). After meeting the requirements with

```bash
sudo apt-get install libyaml-cpp-dev -y
pip3 install --upgrade pip
pip3 install --upgrade cmake
````

I get a lot of compilation errors:

```bash 
mkdir build
cd build
cmake ../
```
<details><summary>cmake output</summary>
<p>

```
CMake Warning (dev) at CMakeLists.txt:21:
  Syntax Warning in cmake code at column 30

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
-- Found CUDNN include: /usr/include
-- Found NVINFER: /usr/lib/aarch64-linux-gnu/libnvinfer.so
-- Found NVINFER include: /usr/include/aarch64-linux-gnu
Eigen DIR: /usr/include/eigen3
install dir:/usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zauberzeug/tkDNN/build
```

</p>
</details>

```bash
make -j 6
[  1%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_upsample.cu.o
[  1%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_elu.cu.o
[  1%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_mish.cu.o
[  2%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_leaky.cu.o
[  3%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_logistic.cu.o
[  4%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_relu_ceiling.cu.o
[  5%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_activation_sigmoid.cu.o
[  5%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_convert.cu.o
[  6%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_deformable_conv.cu.o
[  7%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_fill.cu.o
[  7%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o
[  8%] Building NVCC (Device) object CMakeFiles/kernels.dir/src/kernels/kernels_generated_pooling.cu.o
/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(4): error: incomplete type is not allowed
```

<details><summary>further make output</summary>
<p>

```

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(15): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(19): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(23): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(29): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/ActivationLeakyRT.h(34): error: member function declared with "override" does not override a base class member
[... 182 lines cut ...]

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(15): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(19): error: member function declared with "override" does not override a base class member

/home/zauberzeug/tkDNN/include/tkDNN/pluginsRT/UpsampleRT.h(23): error: member function declared with "override" does not override a base class member

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00002d8f_00000000-6_normalize.cpp1.ii".
Compilation terminated.
CMake Error at kernels_generated_normalize.cu.o.cmake:280 (message):
  Error generating file
  /home/zauberzeug/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_normalize.cu.o

CMakeFiles/kernels.dir/build.make:138: recipe for target 'CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o' failed
make[2]: *** [CMakeFiles/kernels.dir/src/kernels/kernels_generated_normalize.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:906: recipe for target 'CMakeFiles/kernels.dir/all' failed
make[1]: *** [CMakeFiles/kernels.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```

</p>
</details>

mive93, 2021-08-19:
Hi @rodja 

Yes, we know. The problem is TensorRT 8, in which they removed the plugin version tkDNN is build onto.
Therefore, it will take some time for us to change everything. We will do it eventually, but it's still a WIP right now.

mje-nz, 2021-09-27:
Is anyone working on TensorRT 8 support?

mive93, 2021-09-27:
@mje-nz  yes, we are.

mje-nz, 2021-09-27:
@mive93 Cool, anything I can do to help?

Ram-Godavarthi, 2021-11-04:
Hi,
thank you for it.
But I already used yolo-> onnx -> trt 8 to accomplish the task.
It is much better and easy to understand as well.

mive93, 2021-11-24:
TensorRT8 is now supported on [tensorrt8 branch](https://github.com/ceccocats/tkDNN/tree/tensorrt8).
Every model and data type is properly working.

closed 2021-11-24 by mive93