ceccocats/tkDNN #262: Broken darknet export

alexkokh, 2021-09-04:
There's couple issues with exporting weights from darknet:

- the darknet fork is broken, there are unmerged changes in [convolution_kernels.cu](https://git.hipert.unimore.it/fgatti/darknet/-/blob/master/src/convolutional_kernels.cu#L851)
- fopen mode in [darknet.c](https://git.hipert.unimore.it/fgatti/darknet/-/blob/master/src/darknet.c#L455) is wrong (should be "wb") and results in larger files Windows (I haven't tested the Linux version yet). Due to this the exported layer files are corrupt and can't be used.

My solution was to use the current darknet master branch, add the export code to it and change the fopen mode to "wb". Perhaps you may want to fix your darknet fork as well.

MohamedElsaeidy, 2021-09-05:
git clone https://git.hipert.unimore.it/fgatti/darknet.git
i think you should just use this repo for exporting only and make it without any changes , and it would for you pretty good

alexkokh, 2021-09-05:
Have you checked the links in the original post? Those two problems are in the repo you’re referring to.
You can’t even build that code the way it is now and the export is producing corrupt files.

MohamedElsaeidy, 2021-09-05:
for producing larger files , yes it does . i made 2 folders layers , debug and it worked fine.

for convultion_kernals i didn't get issue with it. what model u trying to export

alexkokh, 2021-09-06:
You don't seem to be understanding the problem. It's irrelevant what model is being exported. Binary files shouldn't be written in text mode. It may not make a difference in Linux, however it breaks Windows compatibility.

B1ackg0d, 2021-09-06:
I also encountered the first problem you mentioned, and then I can compile after modifying it.
[image]
But my darknet still cannot generate debug files.

alexkokh, 2021-09-06:
You don’t need the debug files for inference, as long as you have the correctly exported layer files you should be good.
It’s easier to use the up to date darknet code instead of modifying individual files, except for the export code obviously.

ceccocats, 2021-09-29:
> You don't seem to be understanding the problem. It's irrelevant what model is being exported. Binary files shouldn't be written in text mode. It may not make a difference in Linux, however it breaks Windows compatibility.

Hi, I use only Linux and since it has no difference I missed the "b" flag. If you have fixed the issue by adding the correct flag we can update the code. Or you can make a merge request.

Best,
Francesco