ceccocats/tkDNN #272: 'error adding symbols' during "make" in jetson nano

forgiveall, 2021-11-03:
when I make tkDNN in jetson nano, I met unexpected error message as below:

kibum@kibum-desktop:~/tkDNN/build$ make
CMake Warning (dev) at CMakeLists.txt:21:
  Syntax Warning in cmake code at column 30

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found CUDNN: /usr/local/cuda/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda/include
-- Found NVINFER: /usr/lib/aarch64-linux-gnu/libnvinfer.so
-- Found NVINFER include: /usr/include/aarch64-linux-gnu
Eigen DIR: /usr/include/eigen3
install dir:/usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kibum/tkDNN/build
[  1%] Linking CXX shared library libkernels.so
[ 13%] Built target kernels
[ 14%] Linking CXX shared library libtkDNN.so
**/usr/local/cuda/lib64/libcudnn.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status**
CMakeFiles/tkDNN.dir/build.make:667: recipe for target 'libtkDNN.so' failed
make[2]: *** [libtkDNN.so] Error 1
CMakeFiles/Makefile2:750: recipe for target 'CMakeFiles/tkDNN.dir/all' failed
make[1]: *** [CMakeFiles/tkDNN.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

------------------------------------------------------------------------------------------
Please let me know how to solve this error....

Jetpack: 4.5
Cuda: 10.2
cudnn: 8.0.0
opencv: 4.1.1
TensorRT: 7.1.3.0

perseusdg, 2021-11-19:
are you cross compiling ?if so then you can try this guide https://forums.developer.nvidia.com/t/solved-libcudart-so-error-adding-symbols-file-in-wrong-format/71084/3

mive93, 2021-11-25:
Also, which version of cmake are you using?
tkDNN requires cmake>= 3.15 and the default in the Jetpack is way smaller.

forgiveall, 2021-11-26:
@mive93 
I already check my cmake version is 3.18.2 which was upgraded before.
so, I think it is not the reason to make this problem.

forgiveall, 2021-11-26:
@perseusdg 
Thank you for your information.
I checked the cross compiling issue in my jetson nano. but it is not easy to find which folders are conflicted since the issues are discussed with old version in 2years ago.
Anyway, I'll try again :-)