ceccocats/tkDNN #274: Confusion over workflow

lpkoh, 2021-11-23:
HI I would like to run the demo with my own custom yolov4 model.

Is it that I have to make sure that there is a test_nn folder in the build folder before I run, say ./demo?
This test_nn folder, can be generated either by 1) running a ./test_yolo4 in build or 2) exporting weights from darknet and then copying the layer and debug files into a test_nn folder I create myself, right?

perseusdg, 2021-11-23:
Its more along the lines of the second method you mentioned,and you might need to slightly modify the test_yolo4(yolo4.cpp) file to use the cfg and name files used to train your custom network,and also point out  the exact debug layers needed for comparison

lpkoh, 2021-11-23:
Ah. Could I clarify how this works?

So I see when I just go straight to build folder and run test_yolo4, it generates a yolo4 folder that contains a debug and layers folder, containing the binaries. It also generates a yolo4_fp32.rt. With this rt file, I can then proceed to run the demo.

So for my own custom model, e.g. yolo4_x, I would first export the weights using yolo4x.cfg and yolo4x.weights, create a folder called yolo4x in build, copy in the debug and layers from the prevous export. After that, I would need to run ./test_yolo4x to generate an rt file (like ./test_dla34_cnet3d). This test_yolo4x would be generated by copying the yolo4.cpp, and altering things like input, output, cfg path, etc.

I have 2 follow up questions:
1. How do I go from yolo4x.cpp to test_yolo4? Is it like saving the file in a different format or something? I notice I am able to open the yolo4.cpp file in vim, but unable to do so for test_yolo4. In fact I can't open it and have no idea what is actually inside.
2. My retrained model does not alter the architecture, but merely edits the anchors as well as the number/type of classes (I am training off darknet). Would this lead to any changes in the output layers?

perseusdg, 2021-11-23:
1)test_yolo4 is the compiled version of yolo4.cpp , since you are using yolo4x  modifying this [file ](https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4x.cpp) should do the trick 
2) i don't think there would be any changes in  the output layers,just modifying the cfg and name path(the ones you used to train the network should do the trick) in the yolo4x.cpp file