ceccocats/tkDNN #286: terminate called after throwing an instance of 'YAML::ParserException'

Geniuszyd, 2022-04-03:
Loading YAML: yolo4_fp32.rt
terminate called after throwing an instance of 'YAML::ParserException'what(): yaml-cpp: error at line 1, column 8: unknown escape character:Aborted (core dumped)

Could you please tell me why?

mych907, 2022-04-07:
Exact same issue here. Mine was yolo4tiny_fp32.rt
Jetson Nano 4.5.1, TensorRT 7.1.3, CUDA 10.2, OpenCV 4.5.2.

I followed this instruction. (https://github.com/engcang/ros-yolo-sort/tree/master/YOLO_and_ROS_ver)

After exporting the weights and configs to .bin files, .rt file was successfully created.

Trying to run a demo on my sample video, I ran below.
```
./demo yolo4tiny_fp32.rt ~test_video.mp4 y 80 1 1
```

Then the result shows
```
Loading YAML: yolo4tiny_fp32.rt
terminate called after throwing an instance of 'YAML::ParserException'
  what():  yaml-cpp: error at line 1, column 8: unknown escape character: 
Aborted (core dumped)

```
Any help?

Geniuszyd, 2022-04-08:
Maybe you'd better modification the "./demo/demoConfig.yaml" by yourself.

engcang, 2022-04-13:
@mych907 
Hi. You referred my repo. Thanks.
I recently found that this repo (tkDNN) is updated. So to run a demo, you should do like follows:

```
./demo tkDNN/demo/demoConfig.yaml
```

make sure to edit the paths of rt file in the `demoConfig.yaml` file.