ceccocats/tkDNN #287: Build error window

cuongphamduc, 2022-04-15:
I followed the build instructions on windows with respective dependencies but no success. This is the problem I encountered. Can someone help me? I also removed the --compiler-options '-fPIC' option at line 85 in the Cmake file because it doesn't support windows. I attached the error image and my cmake file.
[image]
[CMakeLists.txt](https://github.com/ceccocats/tkDNN/files/8495212/CMakeLists.txt)

perseusdg, 2022-04-15:
I have a couple of questions : 

- are you building it in the release or debug mode ?
- is this the issue when building all test_*.exe or a specific one ?

cuongphamduc, 2022-04-15:
@perseusdg I build on Release mode, and I build all with command :
```
cmake -DCMAKE_BUILD_TYPE=Release -G"Ninja" ..
ninja -j4
```
It seem error occurs when executing build to tkDNN.dll and tkDNN.lib

perseusdg, 2022-04-15:
I think some recent changes to the master branch need some windows specific modific modifications ,till that is sorted out , you can use the tensorrt-8 branch here https://github.com/perseusdg/tkDNN (tested on windows 10/11)

cuongphamduc, 2022-04-15:
Thanks you, I will try.

cuongphamduc, 2022-04-16:
@perseusdg, I have successfully built but I met new error. I have copied cudnn_cnn_infer64_8.dll to the same folder with test_yolo4tiny.exe and tried restart but still error.
[image]

closed 2022-04-16 by cuongphamduc