ceccocats/tkDNN #289: cloud not build cuda engine

Michaelzeyong, 2022-04-18:
I use win10 + cuda10.2 + 2080super, build tkdnn success!
but when I try to transfer .rt files for yolov4 1024*1024, it saia "cloud not build cuda engine". 
yolov4 800*800 success generate fp32 and fp16 .rt files, but only fp16 can use demo.exe. 
I cant solve this error.
anyone else had fix it! 
thank you!

nuriakiin, 2022-09-28:
I have same issue. Cant resolve it. Did you solve it? 
[image]

nuriakiin, 2022-09-29:
I think is about memory or something same. I tried small models and worked. Then try again big model worked.