ceccocats/tkDNN #293: YOLO v7 Support

mkzein, 2022-07-08:
Hello

Any plans to support YOLOv7 :1st_place_medal: ?

seba-eng, 2022-08-12:
I'm going to add Yolov5 and YOLOX to the wish list. Yes, Yolov4 is technically the last official Yolo version, but those other networks are also used in practice.

mkzein, 2022-11-07:
Hello
Any updates on support for YOLOv7?

perseusdg, 2022-11-17:
https://github.com/perseusdg/tkDNN/tree/swish-dev .. this is an untested version of swish for tkdnn , i also enabled it for the darknet parse so it should be relatively easy to use it with yolov7

AGpix, 2023-10-30:
> I tested yolov7 with tkDNN a couple of weeks ago(i lost track of the code,if i get the time , I will rewrite the code and create a pull request over a weekend), it was pretty straightforward to adapt tkDNN for yolov7, the only new layer that you need is swish ,for the tensorrt part you can use a combination of IElementWiseLayer functions and for the cudnn part you can copy the swish kernel from alexyab's darknet repo. To export darknet weights to tkdnn bin files , copy over the lines of code that deal with converting weight files in the hipert darknet repo mentioned somewhere in the docs to latest commit on alexyab's darknet repo and to run it , use the command mentioned in the hipert repo.
> 
> Writing a test_yolov7 function should be more or less along the lines of test_yolov4 , the demo should run out of the box without any issues with yolov7 if you added the swish layer as mentioned above

Hello please can you share in more detail how implemented for yolov7. Thanks

AGpix, 2023-10-30:
> https://github.com/perseusdg/tkDNN/tree/swish-dev .. this is an untested version of swish for tkdnn , i also enabled it for the darknet parse so it should be relatively easy to use it with yolov7

Thanks for the code...which layers did you use for yolov7.

Egorundel, 2024-05-13:
@perseusdg you are tested on yolov7 with this instruction?

https://github.com/perseusdg/tkDNN/blob/swish-dev/docs/exporting_weights.md

Egorundel, 2024-05-13:
@seba-eng @perseusdg @mkzein @AGpix How can this be applied to YOLOv7-8? for a PyTorch implementation? They don't have the files _**model.cfg**_ and _**model.weights**_, but there is only _**model.yaml**_ and _**model.pt**_