ceccocats/tkDNN #295: centernet3d and centertrack3d, too many inferences results showing!

sainttelant, 2022-08-11:
i have converted the cnet3d and centertrack 3d models to *.rt format models, however, when i ran the demos of demotrack and demo3D, i  got lots of detections per frame for both models, i don't know why this happened,  my environment is rtx3090 nvidia card, installed tensorrt8.2GA and cudnn8.X versions, please give me some advices to solve this weired issues!! thanks a lot,
here are results showing :

sainttelant, 2022-08-11:
[image]
[image]

sainttelant, 2022-08-15:
i noticed that this repo hasn't tried to run on RTX3090 tensorrt8.2 yet,,  does anyone have silimar issues with mine?

perseusdg, 2022-08-16:
Hi ,I just tested ctrack and cnet3d 3d demos with tensorrt 8.4 and they seem to be working fine for me on linux.They should work with tensorrt 8.2 also.

sainttelant, 2022-08-17:
@perseusdg , it is so weird, it truely looked like lots of detection results just showing as i detached above,  i have tried to deploy the same program i compiled at jetson xvaier, it looked good , just like official demo showing.  confused !

perseusdg, 2022-08-17:
I can't be exactly sure of what the issue might be ,but i can speculate that it is probably a windows thing , i sometimes get a segmentation fault when running ctrack and cnet3d on windows,.