ceccocats/tkDNN #300: Yolo3Detection segfault on cv::resize - Jetpack 5.0

ckurtz22, 2023-02-05:
Running the Yolo3Detection class in Jetpack 5.0. When I call Yolo3Detection::update, I get a segfault, as shown in gdb below:
[image]

ckurtz22, 2023-02-06:
Seems the issue is that jetpack has installed 2 versions of OpenCV, 4.5.4 and 4.2.0. I'm using tkDNN inside a larger CMake project, and that CMake project is targeting OpenCV 4.2.0 (cv_bridge from ROS Noetic), and tkDNN is targeting the OpenCV 4.5.4 version. Struggling to find a way to make these packages play nice

ckurtz22, 2023-02-06:
Was able to resolve it by manually building and installing opencv and then running the uninstall make target to remove the 4.5.4 files, so only 4.2.0 was left in the system. Now tkDNN targets the 4.2.0 opencv

closed 2023-02-06 by ckurtz22

ChanJoon, 2023-11-25:
your thread was helpful for me. I resolve this issue by manually building cv_bridge with my OpenCV version. (In my case 4.8.0 and 4.2.0(noetic))