ceccocats/tkDNN #304: Any updates if it works with new versions of cuda 11.8?

yemuzi, 2023-08-21:
on 2070 gpu card,
[image]

Cuda failure: out of memory
/root/trt8/src/Conv2d.cpp:177
Aborting...

ckurtz22, 2023-11-05:
I've tested it up to Cuda 12.3, works fine. The memory error is because the tests are written around running it on a Jetson Xavier with unified memory. For normal GPUs you need to comment out this line so when it can just serialize the network without running it: https://github.com/ceccocats/tkDNN/blob/master/src/Conv2d.cpp#L176
And then you can load the network.rt file and run it later on.

vtyw, 2023-11-22:
@ckurtz22 What version of OpenCV did you have to use with CUDA 12?

ckurtz22, 2023-11-23:
> @ckurtz22 What version of OpenCV did you have to use with CUDA 12?

I used the default opencv installation for Ubuntu 20.04, I believe it was 4.1. I also was able to build 4.8 from source and get it working without issue. I have not been using OpenCV with CUDA enabled,